hvoss-tech/AutoForge #1: Limit number of filaments used?

MinchinWeb, 2025-02-20:
Is it possible to limit the number of filaments used?

I have a single AMS, so if I can do this with only 4 colors, that would be preferred. It's fine if a color is re-used.

hvoss-techfak, 2025-02-20:
Currently, you can simply change the number of colors in your csv files to 4, by opening it and removing the other lines. This way the program only uses these colors. 

We also just added Hueforge project support, so you could also add more colors, check your result in Hueforge and then remove the other colors there.

MinchinWeb, 2025-02-24:
I don't know what work would be involved, but if the solver could be limited to 4 (or 8...) colours, that would be helpful from a user perspective. I can trim down my input colour list, but that requires knowing which colours I want to use to start with, and large part of the magic of HueForge (to me) is the blended colours that aren't always obvious, especially if your program is effectively mixing three or four of them to get the displayed shade, rather than pushing to full saturation and switching to the next colour.

hvoss-techfak, 2025-02-24:
I'll reopen this and will see if I can find a good solution for this in the next days. 
Currently you could simply run the model with all of the colors, see which ones are used the most, limit the program to these colors by changing the csv and run again.

hvoss-techfak, 2025-03-01:
We are now on version 1.2.0 and I finally found a good posthoc approach that can limit the colors and color swaps. If you want you can try it out. 

Please note that the library is now in PyTorch and we don't use JAX any longer, but all requirements should be automatically installed with the command:

pip install -U autoforge

closed 2025-03-01 by hvoss-techfak