hvoss-tech/AutoForge #3: Print test

cjbnc, 2025-02-22:
- Source image: https://github.com/cjbnc/AutoForge/blob/main/samples/snow_corgi.jpg
- Filament list: https://github.com/cjbnc/AutoForge/blob/main/samples/dogs.csv

```$ python auto_forge.py   --input_image images/snow_corgi.jpg   --csv_file filaments/dogs.csv   --output_folder outputs   --iterations 80000   --learning_rate 0.01   --layer_height 0.08   --max_layers 25   --background_height 0.4   --background_color "#000000"   --max_size 512
loss = 728.6719, Best Loss = 362.3761:  20%|███████▌                              | 15997/80000 [10:06<41:36, 25.63it/s]Saving intermediate outputs. This can take some time. You can turn off this feature by setting save_interval_pct to 0.
loss = 1586.1084, Best Loss = 362.3761:  40%|██████████████▊                      | 31997/80000 [20:11<30:42, 26.06it/s]Saving intermediate outputs. This can take some time. You can turn off this feature by setting save_interval_pct to 0.
loss = 1468.4768, Best Loss = 362.3761:  60%|██████████████████████▏              | 47997/80000 [30:03<20:45, 25.69it/s]Saving intermediate outputs. This can take some time. You can turn off this feature by setting save_interval_pct to 0.
loss = 1468.6917, Best Loss = 1294.2984:  80%|████████████████████████████▊       | 63997/80000 [39:58<10:45, 24.78it/s]Saving intermediate outputs. This can take some time. You can turn off this feature by setting save_interval_pct to 0.
loss = 1482.4162, Best Loss = 1289.6072: 100%|███████████████████████████████████▉| 79997/80000 [50:08<00:00, 25.21it/s]Saving intermediate outputs. This can take some time. You can turn off this feature by setting save_interval_pct to 0.
loss = 1468.6917, Best Loss = 1289.6072: 100%|████████████████████████████████████| 80000/80000 [50:26<00:00, 26.43it/s]
Project file saved to outputs/project_file.hfp
All outputs saved to outputs
Happy printing!```

- Preview result: https://github.com/cjbnc/AutoForge/blob/main/samples/946394c_corgi_discrete_comp.jpg
- Layer swaps: https://github.com/cjbnc/AutoForge/blob/main/samples/946394c_corgi_swap_instructions.txt
- Loaded into Bambu studio, scaled to 20% in XY (102.4mm) and printed: https://github.com/cjbnc/AutoForge/blob/main/samples/946394c_corgi_printed.jpg

A couple thoughts...

In the run output above, I don't understand why the Best Loss is taking such a big jump sometime between the 60% and 80% checkpoints. I'm guessing maybe I'm running too many iterations and overfitting, perhaps?  Or is there something in teh algorithm that changes the loss scale in the middle?

My filaments for brown and grey are both nearly opaque (TD 0.3) so the program is having a hard time with finding gradients around them. That's not surprising. In HueForge, I would substitute layers of mid-TD orange on black rather than go straight to the opaque brown to get a better result.

I hope this is helpful.

hvoss-techfak, 2025-02-23:
As the Gumbel softmax loss is decreasing there is a point where we reset the best loss once, as otherwise the "best loss" could be a loss loss value but with a very large tau value. A big tau value essentially means that there is a higher chance that two or more colors are mixed in the background to get the result. Currently this is set by the following parameter:
parser.add_argument("--save_max_tau", type=float, default=0.005, help="Tau threshold to save best result")

This helps in forcing a best loss that is more closely aligned with only one color. I'm still not entirely satisfied with this hack, as sometimes the visual result can get actually worse doing this. I think I have some time in the next days to maybe find a better solution for that.

cjbnc, 2025-02-23:
Ah thanks for the explanation. The PR #4 seems to really help with the color matching on my tests. Getting the layers lined up in hueforge per #2 as well gives this kind of result now:
https://github.com/cjbnc/AutoForge/blob/main/samples/b602c9a7_grizzly_af_hf.jpg

I'm running that through my printer now to make sure the print looks similar.

hvoss-techfak, 2025-02-23:
I just found a good solution. We now calculate the validation loss based on the final settings from the discretized image.

commit 8913a1d3 (2025-02-23): Calculate validation loss based on final discretized image. Fixes #3
closed 2025-02-23 by hvoss-techfak