hvoss-tech/AutoForge #8: [Errno 2] No such file or directory

zuriel06, 2025-02-26:
I think I have installed all that is needed and followed everything but I am unable to get it to work. Can you make a video on how to install it and use the autoforge please

[image]

The first think I tried was to find the auto_forge.py and paste it to the error [errno] location but I get another error:

[image]

hvoss-techfak, 2025-02-26:
We just finished adding pypi support. You can now install autoforge by simply running:
pip install autoforge[gpu]
if you have a gpu and
pip install autoforge
If you don't have a gpu

hvoss-techfak, 2025-02-26:
Afterwards you can simply run:
auto_forge --input_image path/to/input_image.jpg --csv_file path/to/materials.csv --output_folder outputs

zuriel06, 2025-02-26:
I ran the pip install autoforge[gpu] but I got this response:

[image]

I am assuming that it wouldn't work with a laptop that had a dedicated GPU then?!

I only tried the pip install autoforge[gpu] because the laptop has a gtx 1080.

After a while it started to download and then this happeded:

Microsoft Windows [Version 10.0.19045.5487]
(c) Microsoft Corporation. All rights reserved.

C:\Users\MSI>pip install autoforge[gpu]
Collecting autoforge[gpu]
  WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ConnectTimeoutError(<pip._vendor.urllib3.connection.HTTPSConnection object at 0x000001B454989C10>, 'Connection to files.pythonhosted.org timed out. (connect timeout=15)')': /packages/d7/3d/f50573c812ddb7577bfde40479b3a4ce51a3a971a8e1566cce4cee5124f3/autoforge-1.1.2-py3-none-any.whl.metadata
  Downloading autoforge-1.1.2-py3-none-any.whl.metadata (8.4 kB)
Requirement already satisfied: numpy>=1.19.0 in c:\python312\lib\site-packages (from autoforge[gpu]) (1.26.4)
Requirement already satisfied: jax>=0.3.10 in c:\python312\lib\site-packages (from autoforge[gpu]) (0.5.1)
Requirement already satisfied: jaxlib>=0.3.10 in c:\python312\lib\site-packages (from autoforge[gpu]) (0.5.1)
Requirement already satisfied: opencv-python>=4.2.0.32 in c:\python312\lib\site-packages (from autoforge[gpu]) (4.11.0.86)
Requirement already satisfied: matplotlib>=3.3.0 in c:\python312\lib\site-packages (from autoforge[gpu]) (3.8.3)
Requirement already satisfied: configargparse>=1.2.3 in c:\python312\lib\site-packages (from autoforge[gpu]) (1.7)
Requirement already satisfied: tqdm>=4.50.0 in c:\python312\lib\site-packages (from autoforge[gpu]) (4.67.1)
Requirement already satisfied: pandas>=1.1.0 in c:\python312\lib\site-packages (from autoforge[gpu]) (2.2.3)
Requirement already satisfied: scikit-image in c:\python312\lib\site-packages (from autoforge[gpu]) (0.25.2)
Requirement already satisfied: scikit-learn in c:\python312\lib\site-packages (from autoforge[gpu]) (1.6.1)
Requirement already satisfied: optax>=0.1.3 in c:\python312\lib\site-packages (from autoforge[gpu]) (0.2.4)
Requirement already satisfied: ml_dtypes>=0.4.0 in c:\python312\lib\site-packages (from jax>=0.3.10->autoforge[gpu]) (0.5.1)
Requirement already satisfied: opt_einsum in c:\python312\lib\site-packages (from jax>=0.3.10->autoforge[gpu]) (3.4.0)
Requirement already satisfied: scipy>=1.11.1 in c:\python312\lib\site-packages (from jax>=0.3.10->autoforge[gpu]) (1.15.2)
Requirement already satisfied: contourpy>=1.0.1 in c:\python312\lib\site-packages (from matplotlib>=3.3.0->autoforge[gpu]) (1.2.0)
Requirement already satisfied: cycler>=0.10 in c:\python312\lib\site-packages (from matplotlib>=3.3.0->autoforge[gpu]) (0.12.1)
Requirement already satisfied: fonttools>=4.22.0 in c:\python312\lib\site-packages (from matplotlib>=3.3.0->autoforge[gpu]) (4.50.0)
Requirement already satisfied: kiwisolver>=1.3.1 in c:\python312\lib\site-packages (from matplotlib>=3.3.0->autoforge[gpu]) (1.4.5)
Requirement already satisfied: packaging>=20.0 in c:\python312\lib\site-packages (from matplotlib>=3.3.0->autoforge[gpu]) (24.0)
Requirement already satisfied: pillow>=8 in c:\python312\lib\site-packages (from matplotlib>=3.3.0->autoforge[gpu]) (10.2.0)
Requirement already satisfied: pyparsing>=2.3.1 in c:\python312\lib\site-packages (from matplotlib>=3.3.0->autoforge[gpu]) (3.1.2)
Requirement already satisfied: python-dateutil>=2.7 in c:\python312\lib\site-packages (from matplotlib>=3.3.0->autoforge[gpu]) (2.9.0.post0)
Requirement already satisfied: absl-py>=0.7.1 in c:\python312\lib\site-packages (from optax>=0.1.3->autoforge[gpu]) (2.1.0)
Requirement already satisfied: chex>=0.1.87 in c:\python312\lib\site-packages (from optax>=0.1.3->autoforge[gpu]) (0.1.89)
Requirement already satisfied: etils[epy] in c:\python312\lib\site-packages (from optax>=0.1.3->autoforge[gpu]) (1.12.0)
Requirement already satisfied: pytz>=2020.1 in c:\python312\lib\site-packages (from pandas>=1.1.0->autoforge[gpu]) (2025.1)
Requirement already satisfied: tzdata>=2022.7 in c:\python312\lib\site-packages (from pandas>=1.1.0->autoforge[gpu]) (2025.1)
Requirement already satisfied: colorama in c:\python312\lib\site-packages (from tqdm>=4.50.0->autoforge[gpu]) (0.4.6)
INFO: pip is looking at multiple versions of jax[cuda12] to determine which version is compatible with other requirements. This could take a while.
Collecting jax[cuda12] (from autoforge[gpu])
  Downloading jax-0.5.0-py3-none-any.whl.metadata (22 kB)
  Downloading jax-0.4.38-py3-none-any.whl.metadata (22 kB)
  Downloading jax-0.4.37-py3-none-any.whl.metadata (22 kB)
  Downloading jax-0.4.36-py3-none-any.whl.metadata (22 kB)
  Downloading jax-0.4.35-py3-none-any.whl.metadata (22 kB)
  Downloading jax-0.4.34-py3-none-any.whl.metadata (22 kB)
  Downloading jax-0.4.33-py3-none-any.whl.metadata (22 kB)
INFO: pip is still looking at multiple versions of jax[cuda12] to determine which version is compatible with other requirements. This could take a while.
  Downloading jax-0.4.31-py3-none-any.whl.metadata (22 kB)
  Downloading jax-0.4.30-py3-none-any.whl.metadata (22 kB)
  Downloading jax-0.4.29-py3-none-any.whl.metadata (23 kB)
  Downloading jax-0.4.28-py3-none-any.whl.metadata (23 kB)
  Downloading jax-0.4.27-py3-none-any.whl.metadata (23 kB)
[... cut ...]
    Uninstalling chex-0.1.89:
      Successfully uninstalled chex-0.1.89
  Attempting uninstall: optax
    Found existing installation: optax 0.2.4
    Uninstalling optax-0.2.4:
      Successfully uninstalled optax-0.2.4
  WARNING: Failed to write executable - trying to use .deleteme logic
ERROR: Could not install packages due to an OSError: [WinError 2] The system cannot find the file specified: 'C:\\Python312\\Scripts\\autoforge.exe' -> 'C:\\Python312\\Scripts\\autoforge.exe.deleteme'

C:\Users\MSI>

[image]

zuriel06, 2025-02-26:
Should I of ran the pip install autoforge?

I am assuming I would have to uninstall something? If yes, I don't know how to do such.

hvoss-techfak, 2025-02-26:
The command should be "autoforge --D:\KNIIGHTS\G.F.F.M 2023\Business\Kayla.jpg --D:\Downloads\AutoForge\BAMBULAB BASIC.csv --D:\KNIIGHTS\G.F.F.M 2023\Business\AutoHueforge Files"

If you are under windows you need to install WSL2 and install autoforge there.

zuriel06, 2025-02-26:
I do apologize for being a Newbie in advance.
I closed off CMD and I would like to know the correct way to run the command you sent. I will leave it installed without the gpu version for now

zuriel06, 2025-02-28:
I attempted it and I placed some extra effort and I got this thus far:

Microsoft Windows [Version 10.0.19045.5487]
(c) Microsoft Corporation. All rights reserved.

C:\WINDOWS\system32>pip install -U autoforge
Requirement already satisfied: autoforge in c:\python312\lib\site-packages (1.1.3)
Requirement already satisfied: numpy>=1.19.0 in c:\python312\lib\site-packages (from autoforge) (1.26.4)
Requirement already satisfied: jax>=0.3.10 in c:\python312\lib\site-packages (from autoforge) (0.4.21)
Requirement already satisfied: jaxlib>=0.3.10 in c:\python312\lib\site-packages (from autoforge) (0.5.1)
Requirement already satisfied: opencv-python>=4.2.0.32 in c:\python312\lib\site-packages (from autoforge) (4.11.0.86)
Requirement already satisfied: matplotlib>=3.3.0 in c:\python312\lib\site-packages (from autoforge) (3.8.3)
Requirement already satisfied: configargparse>=1.2.3 in c:\python312\lib\site-packages (from autoforge) (1.7)
Requirement already satisfied: tqdm>=4.50.0 in c:\python312\lib\site-packages (from autoforge) (4.67.1)
Requirement already satisfied: pandas>=1.1.0 in c:\python312\lib\site-packages (from autoforge) (2.2.3)
Requirement already satisfied: scikit-image in c:\python312\lib\site-packages (from autoforge) (0.25.2)
Requirement already satisfied: scikit-learn in c:\python312\lib\site-packages (from autoforge) (1.6.1)
Requirement already satisfied: optax>=0.1.3 in c:\python312\lib\site-packages (from autoforge) (0.2.2)
Requirement already satisfied: ml-dtypes>=0.2.0 in c:\python312\lib\site-packages (from jax>=0.3.10->autoforge) (0.5.1)
Requirement already satisfied: opt-einsum in c:\python312\lib\site-packages (from jax>=0.3.10->autoforge) (3.4.0)
Requirement already satisfied: scipy>=1.9 in c:\python312\lib\site-packages (from jax>=0.3.10->autoforge) (1.15.2)
Requirement already satisfied: contourpy>=1.0.1 in c:\python312\lib\site-packages (from matplotlib>=3.3.0->autoforge) (1.2.0)
Requirement already satisfied: cycler>=0.10 in c:\python312\lib\site-packages (from matplotlib>=3.3.0->autoforge) (0.12.1)
Requirement already satisfied: fonttools>=4.22.0 in c:\python312\lib\site-packages (from matplotlib>=3.3.0->autoforge) (4.50.0)
Requirement already satisfied: kiwisolver>=1.3.1 in c:\python312\lib\site-packages (from matplotlib>=3.3.0->autoforge) (1.4.5)
Requirement already satisfied: packaging>=20.0 in c:\python312\lib\site-packages (from matplotlib>=3.3.0->autoforge) (24.0)
Requirement already satisfied: pillow>=8 in c:\python312\lib\site-packages (from matplotlib>=3.3.0->autoforge) (10.2.0)
Requirement already satisfied: pyparsing>=2.3.1 in c:\python312\lib\site-packages (from matplotlib>=3.3.0->autoforge) (3.1.2)
Requirement already satisfied: python-dateutil>=2.7 in c:\python312\lib\site-packages (from matplotlib>=3.3.0->autoforge) (2.9.0.post0)
Requirement already satisfied: absl-py>=0.7.1 in c:\python312\lib\site-packages (from optax>=0.1.3->autoforge) (2.1.0)
Requirement already satisfied: chex>=0.1.86 in c:\python312\lib\site-packages (from optax>=0.1.3->autoforge) (0.1.86)
Requirement already satisfied: pytz>=2020.1 in c:\python312\lib\site-packages (from pandas>=1.1.0->autoforge) (2025.1)
Requirement already satisfied: tzdata>=2022.7 in c:\python312\lib\site-packages (from pandas>=1.1.0->autoforge) (2025.1)
Requirement already satisfied: colorama in c:\python312\lib\site-packages (from tqdm>=4.50.0->autoforge) (0.4.6)
Requirement already satisfied: networkx>=3.0 in c:\python312\lib\site-packages (from scikit-image->autoforge) (3.4.2)
Requirement already satisfied: imageio!=2.35.0,>=2.33 in c:\python312\lib\site-packages (from scikit-image->autoforge) (2.37.0)
Requirement already satisfied: tifffile>=2022.8.12 in c:\python312\lib\site-packages (from scikit-image->autoforge) (2025.2.18)
Requirement already satisfied: lazy-loader>=0.4 in c:\python312\lib\site-packages (from scikit-image->autoforge) (0.4)
Requirement already satisfied: joblib>=1.2.0 in c:\python312\lib\site-packages (from scikit-learn->autoforge) (1.4.2)
Requirement already satisfied: threadpoolctl>=3.1.0 in c:\python312\lib\site-packages (from scikit-learn->autoforge) (3.5.0)
Requirement already satisfied: typing-extensions>=4.2.0 in c:\python312\lib\site-packages (from chex>=0.1.86->optax>=0.1.3->autoforge) (4.12.2)
Requirement already satisfied: toolz>=0.9.0 in c:\python312\lib\site-packages (from chex>=0.1.86->optax>=0.1.3->autoforge) (1.0.0)
Requirement already satisfied: setuptools in c:\python312\lib\site-packages (from chex>=0.1.86->optax>=0.1.3->autoforge) (75.8.1)
Requirement already satisfied: six>=1.5 in c:\python312\lib\site-packages (from python-dateutil>=2.7->matplotlib>=3.3.0->autoforge) (1.16.0)

C:\WINDOWS\system32>autoforge --input_image D:\KNIIGHTS\G.F.F.M 2023\Business\Kayla.jpg --csv_file D:\Downloads\AutoForge\BAMBULAB BASIC.csv --output_folder D:\KNIIGHTS\G.F.F.M 2023\Business\AutoHueforge Files
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Python312\Scripts\autoforge.exe\__main__.py", line 4, in <module>
  File "C:\Python312\Lib\site-packages\autoforge\auto_forge.py", line 13, in <module>
    import jax
  File "C:\Python312\Lib\site-packages\jax\__init__.py", line 39, in <module>
    from jax import config as _config_module
  File "C:\Python312\Lib\site-packages\jax\config.py", line 15, in <module>
    from jax._src.config import config as _deprecated_config  # noqa: F401
[... cut ...]
                 [--background_color BACKGROUND_COLOR] [--output_size OUTPUT_SIZE] [--solver_size SOLVER_SIZE]
                 [--decay DECAY] [--visualize] [--perform_gumbal_search PERFORM_GUMBAL_SEARCH]
                 [--perform_pruning PERFORM_PRUNING] [--pruning_max_loss_increase PRUNING_MAX_LOSS_INCREASE]
                 [--save_interval_pct SAVE_INTERVAL_PCT]
autoforge: error: unrecognized arguments: 2023\Business\Kayla.jpg BASIC.csv 2023\Business\AutoHueforge Files

C:\WINDOWS\system32>

hvoss-techfak, 2025-02-28:
Ahh there is a space in the folder. That is probably the error. Please try again with the command:

autoforge --input_image "D:\KNIIGHTS\G.F.F.M 2023\Business\Kayla.jpg" --csv_file "D:\Downloads\AutoForge\BAMBULAB BASIC.csv" --output_folder "D:\KNIIGHTS\G.F.F.M 2023\Business\AutoHueforge Files"

Simply copy the command and execute it.

hvoss-techfak, 2025-03-04:
I'm closing this for now.

closed 2025-03-04 by hvoss-techfak