hvoss-tech/AutoForge #9: Incompatible shapes error with some image sizes

rparrett, 2025-02-27:
Testing 395c9509ac5a55b71b795c87ce9274d5ab17b127

## Reproduction

```
magick -size 254x400 gradient: -rotate 45 -extent 254x400 test_pattern.jpg
PYTHONPATH=src python -m autoforge \
  --input_image=test_pattern.jpg \
  --csv_file=polylite-abs-owned.csv \
  --output_folder=out/test_pattern \
  --solver_size=200 \
  --iterations=100
```

## Output

```
Output folder: out/test_pattern
(200, 127, 3)
Finding best ordering for color bands: Total distance = 95.24: : 40320it [00:00, 115674.86it/s]                                                                           
(1024, 650)
  0%|                                                                                                                                             | 0/100 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/Users/me/src/AutoForge/src/autoforge/__main__.py", line 8, in <module>
    main_cli()
    ~~~~~~~~^^
  File "/Users/me/src/AutoForge/src/autoforge/__main__.py", line 5, in main_cli
    main()
    ~~~~^^
  File "/Users/me/src/AutoForge/src/autoforge/auto_forge.py", line 685, in main
    best_params, _, val_gumbel_keys = run_optimizer(
                                      ~~~~~~~~~~~~~^
        rng_key, target,pixel_height_logits_solver, new_h, new_w, max_layers_value, h_value,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<10 lines>...
        args=args
        ^^^^^^^^^
    )
    ^
  File "/Users/me/src/AutoForge/src/autoforge/auto_forge.py", line 381, in run_optimizer
    params, opt_state, loss = update_step(params, target, tau_height, tau_global, gumbel_keys, opt_state)
                              ~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/me/src/AutoForge/src/autoforge/auto_forge.py", line 235, in update_step
    loss_val, grads = jax.value_and_grad(loss_function)(
                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        params, target, tau_height, tau_global, gumbel_keys,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        h, max_layers, material_colors, material_TDs, background)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/me/src/AutoForge/src/autoforge/auto_forge.py", line 200, in loss_fn
    return jnp.mean((comp - target) ** 2)
                     ~~~~~^~~~~~~~
  File "/Users/me/src/AutoForge/venv/lib/python3.13/site-packages/jax/_src/numpy/array_methods.py", line 1061, in op
    return getattr(self.aval, f"_{name}")(self, *args)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^
  File "/Users/me/src/AutoForge/venv/lib/python3.13/site-packages/jax/_src/numpy/array_methods.py", line 578, in deferring_binary_op
    return binary_op(*args)
  File "/Users/me/src/AutoForge/venv/lib/python3.13/site-packages/jax/_src/numpy/ufunc_api.py", line 179, in __call__
    return call(*args)
  File "/Users/me/src/AutoForge/venv/lib/python3.13/site-packages/jax/_src/numpy/ufuncs.py", line 1544, in subtract
    return lax.sub(*promote_args("subtract", x, y))
           ~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: sub got incompatible shapes for broadcasting: (200, 126, 3), (200, 127, 3).
--------------------
For simplicity, JAX has removed its internal frames from the traceback of the following exception. Set JAX_TRACEBACK_FILTERING=off to include these.
```

rparrett, 2025-03-02:
This still reproduces after the torch rewrite with the following outputL

```
/Users/me/src/AutoForge/src/autoforge/Loss/LossFunctions.py:60: UserWarning: Using a target size (torch.Size([200, 127, 3])) that is different to the input size (torch.Size([200, 126, 3])). This will likely lead to incorrect results due to broadcasting. Please ensure they have the same size.
  mse_loss = F.mse_loss(comp, target) / 1000.0
  0%|                                                                                                                           | 0/100 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/Users/me/src/AutoForge/src/autoforge/__main__.py", line 8, in <module>
    main_cli()
    ~~~~~~~~^^
  File "/Users/me/src/AutoForge/src/autoforge/__main__.py", line 5, in main_cli
    main()
    ~~~~^^
  File "/Users/me/src/AutoForge/src/autoforge/auto_forge.py", line 247, in main
    loss_val = optimizer.step(record_best=i % 3 == 0)
  File "/Users/me/src/AutoForge/src/autoforge/Modules/Optimizer.py", line 165, in step
    loss = loss_fn(
        self.params,
    ...<9 lines>...
        add_penalty_loss=False,
    )
  File "/Users/me/src/AutoForge/src/autoforge/Loss/LossFunctions.py", line 36, in loss_fn
    return compute_loss(
        material_assignment=params["global_logits"],
    ...<5 lines>...
        add_penalty_loss=add_penalty_loss,
    )
  File "/Users/me/src/AutoForge/src/autoforge/Loss/LossFunctions.py", line 60, in compute_loss
    mse_loss = F.mse_loss(comp, target) / 1000.0
               ~~~~~~~~~~^^^^^^^^^^^^^^
  File "/Users/me/src/AutoForge/venv/lib/python3.13/site-packages/torch/nn/functional.py", line 3884, in mse_loss
    expanded_input, expanded_target = torch.broadcast_tensors(input, target)
                                      ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/Users/me/src/AutoForge/venv/lib/python3.13/site-packages/torch/functional.py", line 76, in broadcast_tensors
    return _VF.broadcast_tensors(tensors)  # type: ignore[attr-defined]
           ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
RuntimeError: The size of tensor a (126) must match the size of tensor b (127) at non-singleton dimension 1
```

rparrett, 2025-03-04:
I think this got fixed by 3d40e78117deeecbd5542d3f5bcc9ddcf3b1adf9.

hvoss-techfak, 2025-03-04:
Yeah I'm currently working on some color issues and fixed the size along the way. Please don't use the nightly version right now at it is broken, but as soon as I'm done fixing things I will make a new release which will include the size mismatch fix.

commit c9cea48e (2025-03-04): I've now printed all filaments I have, recorded them and ran my tests on it. I still have to print the actual files to compare the result, but the color should now be even closer to the actual printed output. There is still a small absolut error with my polynomial function, but I would have to think of a better function for it to fit even better.
Also fixed #9 and made some other small changes
closed 2025-03-04 by hvoss-techfak