hvoss-tech/AutoForge #26: Fireflies

cjbnc, 2025-03-15:
I've been using your `generate_stl` imported into my own code and also noticed what you called fireflies in your commit message. My quick solution was to double the mesh in both directions. That effectively makes each point into a square of 4 points all at the same height.  The slicer can usually print those at the right level, so long as the resolution isn't too high. The other problem is that it doubles the number of triangles generated. I find that to be acceptable although others may not.

Here's the snippet of my code:

```
        # double the mesh to get square-ish heights at each pixel
        height_map = np.zeros((2*idxmap.size[1], 2*idxmap.size[0]), dtype=np.float32)
        amask = None
        if use_alpha:
            amask = np.zeros((2*idxmap.size[1], 2*idxmap.size[0]), dtype=np.float32)
        for i in range(idxmap.size[0]):
            for j in range(idxmap.size[1]):
                idx=idxmap.getpixel([i,j])
                v = ncol-idx-1 if reverse else idx
                height_map[2*j,2*i] = v*lh
                height_map[2*j,2*i+1] = v*lh
                height_map[2*j+1,2*i] = v*lh
                height_map[2*j+1,2*i+1] = v*lh
                if use_alpha:
                    av = alpha.getpixel([i,j])
                    amask[2*j,2*i] = av
                    amask[2*j,2*i+1] = av
                    amask[2*j+1,2*i] = av
                    amask[2*j+1,2*i+1] = av
        print("Writing verticies to", stl_model)
        # max_x_y = num pixels * scale
        generate_stl(height_map, stl_model, 0.48, size*scale, amask)
```

hvoss-techfak, 2025-03-16:
Hey, 
thanks for the suggestion. 
your solution would probably work, but our mesh is already above one million polygons on average which slows down most slicers. Right now I use a smoothness penalty for the learned height map which removes most of them, but maybe I'll add your code and then use trimesh to optimize the mesh again to reduce the problem even further.

hvoss-techfak, 2025-03-16:
Yeah I just tried it and if I double the size, the STL becomes around 250mb and crashes orcaslicer. So for now I'll just keep my original code and maybe add it again if I have some more robust mesh decimation functions

closed 2025-03-16 by hvoss-techfak