hvoss-tech/AutoForge #30: windows: `cv2` chokes on utf in image paths

rparrett, 2025-05-14:
```
PS C:\Users\Rob\Desktop\src\AutoForge> cp .\images\cat.jpg テスト\
PS C:\Users\Rob\Desktop\src\AutoForge> autoforge --input_image="テスト/cat.jpg" --csv_file=all-abs-owned.csv
Using device: cpu
[ WARN:0@3.557] global loadsave.cpp:268 cv::findDecoder imread_('πâåπé╣πâê/cat.jpg'): can't open/read file: check file path/integrity
Computed solving pixel size: 750
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\Rob\AppData\Local\Programs\Python\Python312\Scripts\autoforge.exe\__main__.py", line 7, in <module>
  File "C:\Users\Rob\Desktop\src\AutoForge\src\autoforge\auto_forge.py", line 310, in main
    if img.shape[2] == 4:
       ^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'shape'
```

It seems like a workaround might be to use `nd` to read/write files to/from the disk instead, which seems to work on my end but feels janky. https://stackoverflow.com/a/57872297

This doesn't seem to be an issue on macOS.

rparrett, 2025-05-14:
More info, with a slightly nicer workaround: https://github.com/opencv/opencv/issues/4292#issuecomment-2266019697

hvoss-techfak, 2025-05-14:
Thanks for the bug report. This should now be fixed.

closed 2025-05-14 by hvoss-techfak