hvoss-tech/AutoForge #33: Add instructions for local development

rparrett, 2025-05-14:
This is probably obvious to seasoned python devs, but I think some instructions for the intended dev workflow would be helpful

- It seems that one needs to invoke `pip install -e .` to install in "edit mode" in order for local changes to take effect immediately when running with `autoforge`.
- I think there's a manual step after installing with pip to get a version of torch with cuda support installed into the venv?
- How do you build a final package with cuda support?

related: #31

hvoss-techfak, 2025-05-14:
I just added some instructions to the readme. 
You can't really build a final package with cuda support, as pip simply installs all requirements and sometimes misses the cuda requirements. Using the commands I listed is probably the easiest way to get cuda support for the project.

closed 2025-05-14 by hvoss-techfak

rparrett, 2025-05-14:
Thanks! This looks quite helpful. After being away from AutoForge for a bit I was scratching my head a bit trying to remember how to work on it.