hvoss-tech/AutoForge #34: Using .json filament libraries.

xJylin, 2025-05-22:
Hey,

Did you consider implementing a --json_file parameter to allow loading filament lists in json format?

This would greatly increase usability because it would allow users to use our personal libraries of filament which apparently can't be exported in HueForge as CSV
However the json for these libraries still exist under %APPDATA%\HueForge\Filaments\personal_library.json

hvoss-techfak, 2025-05-24:
Hey, thanks for the suggestion. I just added a new parameter --json_file which can be used to load the hueforge json file.

closed 2025-05-24 by hvoss-techfak

hvoss-techfak, 2025-05-24:
The Huggingface Space now also support json file imports: https://huggingface.co/spaces/hvoss-techfak/Autoforge